Safecast/onyxfirmware #66: Doc updates

robouden, 2016-01-11:
It would be helpful for us to write up a page on Safecast site that is a flowchart of options kind of thing. Right now there are 20+ different links that people have to click around to and even then their problem might not be solved. We need something like nano.safecast.org that is one page with everything, and has very clear steps for different issues with the assumption that people know nothing, not the assumption that they are tech

elafargue, 2016-01-11:
I can take this one if you wish - there is good (?) documentation on the Wizkers page for the Onyx, and we can use this as a base, or at least point to it.

As @seanbonner mentioned, we need a 'dummy proof' solution for upgrading old Onyx firmware versions on the Mac, since the current solution involves a couple of commands on the command line which might scare off a couple of users.

robouden, 2016-01-12:
Ed,

yes, go ahead. You are good in documentation.

rob

Regards,
Rob Oudendijk Yuka Hayashi
http://yr-design.biz   http://oudendijk.biz  http://about.me/robouden
tel +81 80-22605966  Skype: robouden  Facebook:robouden
<http://on.fb.me/QeKw2P> twitter:robouden <http://bit.ly/RAiSTC>